CMB27/ModbusRTUMaster #4: Problem When Using With Pi Pico

mohdrais, 2023-10-17:
How to overcome this warning?
c:\users\user\.platformio\packages\framework-arduinopico\arduinocore-api\api\hardwareserial.h:49:71: warning: conversion from 'long unsigned int' to 'uint8_t' {aka 'unsigned char'} changes value from '1043' to '19' [-Woverflow]

CMB27, 2023-10-27:
I need more information to be able to help you.
What core/board are you using? ([earlephilhower/arduino-pico](https://github.com/earlephilhower/arduino-pico)?)
What code are you compiling? Is it one of the examples, or is it your own code? If it is your own code or a modified example, please post the code in full here.
Also, does the code work despite the warning?

Thanks,

mohdrais, 2023-10-31:
I am using raspberry pi pico w [earlephilhower/arduino-pico]. Below is the code (my own code). it is not working despite abble to upload.
------------------------------------------------------------------------------------------------------
#include <ModbusRTUMaster.h>

ModbusRTUMaster MyModbus(Serial1);
uint16_t Reading[8];
bool Suksed;
uint8_t Ans;
void setup() {

  Serial1.setTX(0);
  Serial1.setRX(1);
  Serial1.begin(4800);
  Serial.begin(9600);
}

void loop() {
  delay(100);
  Ans = MyModbus.getExceptionResponse();
  Suksed = MyModbus.readHoldingRegisters(1,0,Reading,8);
  delay(100);
  if (Suksed)
  {
    Serial.println("Success");
    Serial.println(Ans);
  }
  else
  {
    Serial.println("Failed");
    Serial.println(Ans);
  }
  delay(1000);
}
------------------------------------------------------------
this is the output and warnings:
------------------------------------------------------
In file included from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.0\cores\rp2040/api/HardwareSerial.h:2,
                 from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.0\cores\rp2040/SerialUSB.h:24,
                 from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.0\cores\rp2040/Arduino.h:117,
                 from C:\Users\User\AppData\Local\Temp\arduino\sketches\B8A7FDD21D1E4361FD30FA175B71CBE7\sketch\sketch_oct31a.ino.cpp:1:
c:\users\user\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.6.0\arduinocore-api\api\hardwareserial.h:49:71: warning: conversion from 'long unsigned int' to 'uint8_t' {aka 'unsigned char'} changes value from '1043' to '19' [-Woverflow]
   49 | #define SERIAL_8N1           (SERIAL_STOP_BIT_1 | SERIAL_PARITY_NONE  | SERIAL_DATA_8)
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ModbusRTUMaster\src/ModbusRTUMaster.h:22:48: note: in expansion of macro 'SERIAL_8N1'
   22 |     void begin(uint32_t baud, uint8_t config = SERIAL_8N1);
      |                                                ^~~~~~~~~~
In file included from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.0\cores\rp2040/api/HardwareSerial.h:2,
                 from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.0\cores\rp2040/SerialUSB.h:24,
                 from C:\Users\User\AppData\Local\Arduino15\packages\rp2040\hardware\rp2040\3.6.0\cores\rp2040/Arduino.h:117,
                 from c:\Users\User\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.h:7,
                 from c:\Users\User\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.cpp:1:
c:\users\user\appdata\local\arduino15\packages\rp2040\hardware\rp2040\3.6.0\arduinocore-api\api\hardwareserial.h:49:71: warning: conversion from 'long unsigned int' to 'uint8_t' {aka 'unsigned char'} changes value from '1043' to '19' [-Woverflow]
   49 | #define SERIAL_8N1           (SERIAL_STOP_BIT_1 | SERIAL_PARITY_NONE  | SERIAL_DATA_8)
      |                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~
c:\Users\User\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.h:22:48: note: in expansion of macro 'SERIAL_8N1'
   22 |     void begin(uint32_t baud, uint8_t config = SERIAL_8N1);
      |                                                ^~~~~~~~~~
Sketch uses 315204 bytes (15%) of program storage space. Maximum is 2093056 bytes.
Global variables use 71448 bytes (27%) of dynamic memory, leaving 190696 bytes for local variables. Maximum is 262144 bytes.
Resetting COM4
Converting to uf2, output size: 667136, start address: 0x2000
Scanning for RP2040 devices
Flashing F: (RPI-RP2)
Wrote 667136 bytes to F:/NEW.UF2
-------------------------------------------------------
Thank you

CMB27, 2023-10-31:
The [earlephilhower/arduino-pico](https://github.com/earlephilhower/arduino-pico) core uses different values for the Serial configuration than the [Arduino AVR](https://github.com/arduino/ArduinoCore-avr) core.

The Serial configuration values in the AVR core never exceed 255 (8 bits). This library was primarily written for AVRs, so the variables that hold this configuration are of type `uint8_t`. The Serial configuration values in the earlephilhower RP2040 core sometimes exceed 255, so a larger variable type is needed to store the configuration without causing issues.

I can change the configuration variable datatypes to `uint16_t` and that should fix the issue you are having.

CMB27, 2023-10-31:
The changes have been implemented. Please test and let me know if it works. Please use the latest version of the library from GitHub. I have not added a new tag, so these changes are not currently available through the library manager in the Arduino IDE. I don't know if they will show up in PlatformIO either.

mohdrais, 2023-11-01:
YES it seems to work, since there is no warning while compiling and
uploading., both on Arduino and PlatformIO. I have not yet tested using
RS485 (Max3485 -TTL). THANK YOU

CMB27, 2023-11-12:
I have tested the library with an [Arduino Nano RP2040 Connect](https://store.arduino.cc/products/arduino-nano-rp2040-connect) board and the [earlephilhower/arduino-pico](https://github.com/earlephilhower/arduino-pico) core.

I used a modified version of [ModbusRTUMasterExample](https://github.com/CMB27/ModbusRTUMaster/tree/main/examples/ModbusRTUMasterExample).
I tested with an RS-485 transceiver ([LT1785](https://www.analog.com/media/en/technical-documentation/data-sheets/LT1785-1785A-1791-1791A.pdf)).
I used an [Arduino Uno Rev3](https://store.arduino.cc/products/arduino-uno-rev3-smd) running [ModbusRTUSlaveExample](https://github.com/CMB27/ModbusRTUSlave/blob/main/examples/ModbusRTUSlaveExample/ModbusRTUSlaveExample.ino) and equipped with an RS-485 transceiver as the slave/server device.

It worked well when using [HardwareSerial](https://github.com/earlephilhower/ArduinoCore-API/blob/ece6e68f29c6f406a4434659bcbcfe558baaa3a9/api/HardwareSerial.h).
I was not able to get the library working correctly with [SoftwareSerial](https://github.com/earlephilhower/arduino-pico/blob/master/cores/rp2040/SoftwareSerial.h).

closed 2023-11-16 by CMB27