CMB27/ModbusRTUMaster #6: Problem with ESP8266

cturqueti, 2023-11-09:
i try work with ESP8266 an ai have a problem:

c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.cpp:43:6: error: no declaration matches 'void ModbusRTUMaster::begin(long unsigned int, uint8_t)'
   43 | void ModbusRTUMaster::begin(unsigned long baud, uint8_t config) {
      |      ^~~~~~~~~~~~~~~
In file included from c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.cpp:1:
c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.h:22:10: note: candidate is: 'void ModbusRTUMaster::begin(uint32_t, uint8_t)'
   22 |     void begin(uint32_t baud, uint8_t config = SERIAL_8N1);
      |          ^~~~~
c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.h:12:7: note: 'class ModbusRTUMaster' defined here
   12 | class ModbusRTUMaster {
      |       ^~~~~~~~~~~~~~~
c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.cpp:289:6: error: no declaration matches 'void ModbusRTUMaster::_calculateTimeouts(long unsigned int, uint8_t)'
  289 | void ModbusRTUMaster::_calculateTimeouts(unsigned long baud, uint8_t config) {
      |      ^~~~~~~~~~~~~~~
In file included from c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.cpp:1:
c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.h:57:10: note: candidate is: 'void ModbusRTUMaster::_calculateTimeouts(uint32_t, uint8_t)'
   57 |     void _calculateTimeouts(uint32_t baud, uint8_t config);
      |          ^~~~~~~~~~~~~~~~~~
c:\Users\Carlos\Documents\Arduino\libraries\ModbusRTUMaster\src\ModbusRTUMaster.h:12:7: note: 'class ModbusRTUMaster' defined here
   12 | class ModbusRTUMaster {
      |       ^~~~~~~~~~~~~~~

exit status 1

Compilation error: exit status 1

CMB27, 2023-11-12:
Can I get some more information?

- What core and board are you using?
- What sketch are you trying to compile?  
If it is an unmodified example sketch, please let me know which one.  
If it is a custom sketch, please post it here in full.

cturqueti, 2023-11-17:
I removed the config variable and it worked on the ESP32, I think that on the ESP32 there is no serial configuration SERIAL8N1

CMB27, 2023-11-18:
As per [HardwareSerial](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h) from the Espressif [arduino-esp32](https://github.com/espressif/arduino-esp32/tree/master) core, there is a `SERIAL_8N1`, however the defined value would require a 32 bit variable to store it. Up until recently I have used 8 bit variables to store/transfer the configuration, which is sufficient for an AVR, but not an RP2040 or ESP32, therefore I have changed all the variables that handle configuration from `uint8_t` to `uint32_t`. Using the latest version of this library may solve your issue.

cturqueti, 2023-11-20:
It worked, thank you.

closed 2023-11-21 by CMB27