CMB27/ModbusRTUMaster #9: Problem !!

Invi0, 2024-01-27:
Hello. I have a problem with your library. The code only works for me on arduino uno and arduinlo nano boards. (Same MCU). On Leonardo and arduino Mega boards, the library is broken.

CMB27, 2024-01-27:
I just tested this library on an Arduino Leonardo and an Arduino Mega 2560, and didn't encounter any problems.
I'm going to need some more information to be able to help you.

- What version of the library are you using? 1.0.5 is the latest.
- What code/sketch are you running?
  - Is it one of the example programs? If so, which one?
  - Is it your own code or a modified example? If so, please post it here; copy and paste the entire text of the sketch.
- What are you using as a slave/server device to test the library?

Invi0, 2024-01-27:
I'm using your sample code with buttons and buttons for testing. everything works great on arduino uno and arduino nano. On this basis, I built a device communicating with ten slave units. Everything failed when trying to incorporate mega. so I tried a purely basic sketch of the slave device and the master device themselves on the example code. unfortunately the sketch failed and I don't know why. according to the oscilloscope, neither device is transmitting. Tested even without max485 modules on direct rx/tx Tx/Rx. if I use only a simple sketch for sending data, it works on the given pins. I don't know what could be wrong. Could it be an IDE compiler error ? I am using the latest version of the library.

CMB27, 2024-01-27:
What pins are you using for RX and TX?

Invi0, 2024-01-27:
For Pins softwareserial, I used the default pins according to the sketch, then tried it on pins 3, 4, 5, then I switched to using the hw port on the megabyte. By default, I use the classic serial1 and for that I have already started it somehow, but only if I use only one slave, if I modify the code and add another slave like with nano, the other one doesn't work for me. at the same time, I now have a string of 15 slave devices on the nano. however, hw serial now at least communicates with one slave. but I don't know why the serial software doesn't work. , unfortunately it still doesn't work on leonard even when using the HW serial port.

CMB27, 2024-01-27:
SoftwareSerial is particular on what pins can be used on the Arduino Mega and the Arduino Leonardo. The details can be found in the [Arduino Docs for the SoftwareSerial library](https://docs.arduino.cc/learn/built-in-libraries/software-serial/).

The [ModbusRTUMasterExample](https://github.com/CMB27/ModbusRTUMaster/blob/main/examples/ModbusRTUMasterExample/ModbusRTUMasterExample.ino) sketch is only setup to talk to one slave device. I'm guessing you made some modifications. Can you post your sketch here?

closed 2024-02-10 by CMB27