CMB27/ModbusRTUMaster #11: How to Not Use Depin？

654955321, 2024-03-10:
[image]
In my circuit design, there is no DEPIN, and R485 automatically switches between sending and receiving. How to modify and use your library?

654955321, 2024-03-11:
I changed it to "NO_DE_PIN", but the problem still exists. The MCU I use is STM32F103, and I use PlatformIO to compile it. The error prompts that the "avr/interrupt.h" header file cannot be found. Can this library be used normally on stm32?
[image]

CMB27, 2024-03-11:
It looks like you are having issues with `SoftwareSerial` not the DE pin.
I have not tested this library with STM32 microcontrollers, so I don't know how or if this library will work on that platform.

If you share your code, I may be able to help a little more, but I can't guarantee it will work.

closed 2024-03-23 by CMB27