CMB27/ModbusRTUMaster #13: STM32 in PlatformIO

654955321, 2024-03-27:
Hello, I had asked an issue. due to not testing and not replying to you due to some other things in the meantime, the issue was closed, sorry.

Here is my code:

```
#include <Arduino.h>
#include <SoftwareSerial.h>
#include <ModbusRTUMaster.h>

const uint8_t rxPin = PB11;
const uint8_t txPin = PB10;
HardwareSerial modbus_serial(rxPin, txPin);
ModbusRTUMaster modbus(modbus_serial, NO_DE_PIN); // serial port, driver enable pin for rs-485 (optional)

void setup()
{
  modbus.begin(115200); // baud rate, config (optional)
}

void loop()
{
  uint16_t holdingRegisters[2] = {0x0100};
  modbus.readHoldingRegisters(1, 1, holdingRegisters, 1);
  delay(1000);
}
```
I use a 485 to TTL module to receive outgoing messages through my computer. It sends fine, but I'm still having some problems. The above code sends the value "01 03 00 01 00 01 D5 CA"; and the value I need to receive normally is "01 03 01 00 00 01 85 F6". How can I change it, I hope to get your help.

CMB27, 2024-03-28:
In `loop`, try this.
```C++
uint16_t holdingRegisters[2]; // This doesn't need to be set to a value. readHoldingRegisters writes to this array; it doesn't read from it.
modbus.readHoldingRegisters(1, 0x0100, holdingRegisters, 1);
delay(1000);
```

654955321, 2024-03-31:
It works fine now, thanks for your help。
[image]

closed 2024-04-01 by CMB27