CMB27/ModbusRTUMaster #17: Error with ESP8266 Compiling on Arduino IoT Cloud

onostec, 2024-08-28:
Could you please check out this error I got while compiling on Arduino Iot Cloud

/usr/local/bin/arduino-cli compile --fqbn esp8266:esp8266:generic --build-cache-path /tmp --output-dir /tmp/2572720623/build --build-path /tmp/arduino-build-A8648914555524E4D21267121210C4A7  /tmp/2572720623/Untitled_aug28b

/home/builder/Arduino/libraries/modbusrtumaster_1_0_5/src/ModbusRTUMaster.cpp: In member function 'void ModbusRTUMaster::begin(long unsigned int, uint32_t)':
/home/builder/Arduino/libraries/modbusrtumaster_1_0_5/src/ModbusRTUMaster.cpp:67:40: error: invalid conversion from 'uint32_t {aka unsigned int}' to 'SerialConfig' [-fpermissive]
     _hardwareSerial->begin(baud, config);
                                        ^
In file included from /home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/Arduino.h:263:0,
                 from /home/builder/Arduino/libraries/modbusrtumaster_1_0_5/src/ModbusRTUMaster.h:7,
                 from /home/builder/Arduino/libraries/modbusrtumaster_1_0_5/src/ModbusRTUMaster.cpp:1:
/home/builder/.arduino15/packages/esp8266/hardware/esp8266/2.5.0/cores/esp8266/HardwareSerial.h:78:10: error:   initializing argument 2 of 'void HardwareSerial::begin(long unsigned int, SerialConfig)' [-fpermissive]
     void begin(unsigned long baud, SerialConfig config)
          ^

Error during build: exit status 1

CMB27, 2024-08-29:
Huh, In whatever core you are using, it looks like they changed the serial configuration (like `SERIAL_8N1`) to be of type `SerialConfig` rather than an integer type.

What core/board are you using?

onostec, 2024-08-29:
In the Arduino IoT cloud, I only selected the generic ESP8266 and attempted compiling.