CMB27/ModbusRTUMaster #21: Feature Request: Expose the underlying ModbusRTUComm object

EmperorArthur, 2025-05-10:
I would really appreciate it if this library exposed the underlying `ModbusRTUComm` object.  Perhaps via something like a `getRTUComm()` function.

# Rationale
I am transitioning to using this library from old code which wrote directly to the stream.  As an intermediate step, I converted the code to using `ModbusRTUComm`.  The code also has functionality allowing a user to send arbitrary Modbus commands.

To work with the current code, and support the arbitrary command functionality, I have to keep track of both a `ModbusRTUMaster` object, and a separate `ModbusRTUComm`  object.

CMB27, 2025-05-10:
Your application is fairly unique. This library was written to cover the most common use cases of Modbus.
My answer to your request is no.

You are free to fork this library or copy code from it and make any modification you desire.

closed 2025-05-10 by CMB27

EmperorArthur, 2025-05-10:
What are your objections to allowing someone to pass their own object in?  C++ [lifetime rules](https://en.cppreference.com/w/cpp/language/reference_initialization) mean you only need to change the `ModbusRTUComm` to a reference and don't have to manage a separate object.

I'll submit a PR to show you what I mean.

CMB27, 2025-05-11:
I don't see how this feature is generally useful. I understand that it will help you with your code, but I don't really know what your code does, or if this feature is something that other people are likely find useful. I think there are other ways for you to achieve your goal here. You could make a customized version of this library, tailored to your use case. You could copy the functions you use from this library and put them into your own code. If you want me to do some work to add this feature to this library, or approve your pull request, you need to convince me of the value of this feature. At present I am not convinced.